flucto-gmbh/motion-sensor-box #10: No more SSH connection after performing the software system setup

ahaselsteiner, 2021-12-09:
After I conducted the software system setup (see https://github.com/flucto-gmbh/motion-sensor-box/blob/main/doc/system-setup.md) and rebooted, I lost the SSH connection.

We tried to connect both via flucto.tech SSH and via local WLAN. In both cases no connection was possible.

After reading the debugging page (https://github.com/flucto-gmbh/motion-sensor-box/blob/main/doc/debugging.md) I opened the cybercube to check the LEDs. 

* I did not see Raspberry's green light
* StromPi's blue light was on
* A white light from sensehat or paper hat (unsure) was flashing

This movie shows the situation.

https://user-images.githubusercontent.com/11452055/145377978-7b8cee79-394b-487f-b2ac-1f74feb453d4.MOV

I am not sure whether this is normal or my nose tricked me, but I smelled a slight burn scent when opening the cube.

ahaselsteiner, 2021-12-09:
I connected RaspberryPi via USB to my computer to supply it with power. No LED lights up. 
@k323r does that mean that the RaspberryPi is broken? Or could it be something with the SD card (a software problem) and I should flash it again?

[image]

ahaselsteiner, 2021-12-09:
It seems that indeed the RaspberryPi is broken. A new RaspberrPi with the same SD card booted. The green LED lights and I can connect to it with SSH (not over the flucto.tech domain, but locally):

![image](https://user-images.githubusercontent.com/11452055/145444563-77c6827d

https://user-images.githubusercontent.com/11452055/145444668-9eaeed1c-cd0b-4cf7-9abf-55b198435219.MOV

-86d2-4291-ae80-cb4b6d934223.png)

ahaselsteiner, 2021-12-09:
I performed two additional steps:

1) I added the StromPi: Everything still worked as expected.

https://user-images.githubusercontent.com/11452055/145447852-e2002b67-7746-4f24-b22f-26e3fb93d536.MOV

2) I added the remaining stack (lora + breadboard + sensehat + epaper): The raspi did not turn on anymore, its light did not flash. Instead I noticed a burning scent. I think some issue in the remaining stack causes raspberrypi to fail.

ahaselsteiner, 2021-12-14:
On the LoRa-Board there is 0 Ohm resistance between some pins where I did not expect it:
Ground + P2 (blue 13 on Raspberry Pi; https://github.com/flucto-gmbh/motion-sensor-box/blob/main/doc/MSB_pinout.png)
Ground + P3 (blue 15 on Raspberry Pi)

ahaselsteiner, 2021-12-14:
Im checking different combinations of Raspi + USB power + something from upper electronics stack:
Raspi + e-Paper: ok
Raspi + Lora: ok
Raspi + Sensehat: ok
Raspi + breadboard that has two wires connected to strompi: ok
Raspi + StromPi + two wires connected to breadboard: ok
Raspi + StromPi + Lora: ok
Raspi + StromPi + Lora + Breadboard: ok
Raspi + StromPi + Lora + Breadboard + Sensehat: ok
Raspi + StromPi + Lora + Breadboard + Sensehat + epaper hat (full stack): ok

ahaselsteiner, 2021-12-14:
Although RaspberryPi is on, I cannot access it via SSH anymore.

Possibly, there is something wrong with the software on the SD card? Has setting the system configuration broken something (https://github.com/flucto-gmbh/motion-sensor-box/blob/main/doc/system-setup.md)?

ahaselsteiner, 2021-12-14:
I was expecting to short the Pi now but it did not happen after step 2):
1) StormPi power from battery + full stack: ok

2) Power off (plug out), wait 10 seconds until sense hat light fades, power on (plug in): ok
3) Power off (plug out), wait 2 seconds, power on (plug in): no more light on Raspi (possibly some burnt scent)
4) Power off (plug out), wait 10 seconds, power on (plug in): ok (possibly some burnt scent)
5) Power off: now sense hat and StromPi's lights immediately went out, wait 10 seconds, power on (plugin): no more lights on Raspi and on sensehat (strong burnt scent)
6) Power off, wait 10 seconds, power on : no more lights on Raspi
7) Try to power Raspi with USB: Does not work anymore, it's burnt 

@k323r I think the problem is caused by StromPi.

Either:
a) StromPi needs to be configured differently
b) We are not allowed to plug off and plug on power with StromPi (at least with the current configuration)
c) The particular StormPi piece I am using has a failure

ahaselsteiner, 2022-02-18:
I renamed the issue to "RaspberryPi is burned by StromPi (?)". @k323r experienced similar problems about a week ago.

pansegan, 2022-05-11:
I have burned a raspberry by removing the plug between cybercube and power adapter (maybe there was a loose contact while deconecting the electricity)

k323r, 2023-03-23:
this is caused by the strompi, whenever the condensators are connected to and the strompi is powered on again after power off within 30 s.