flucto-gmbh/motion-sensor-box #26: Calculate attitude angles correctly when the box is oriented as certain problematic orientations

ahaselsteiner, 2022-02-18:
We currently experience problems if the box is tilted almost 90° around the x-axis when we want to measure the orientation around the original y-axis.

The following link (private) contains a dataset that shows the problem: https://fluctogmbh.sharepoint.com/:i:/s/flucto2/EXotWqDMpSVHsCa6zRgjo8QBKWRCK2Y-sAazY3EN9A2WSA?e=gQjEQN 

This is related to issue #18 (we should consider including the problematic orientations in the test protocol).

k323r, 2022-05-09:
The inertial measurement unit includes a "digital motion processor (DMP)", a fpga that can be used to automatically estimate the attitude of the imu in the reference coordinate system (aka earth). If the DMP works, the current issued would be resolved. I plan to refactor the msb-imu service to include the DMP.

k323r, 2023-03-26:
duplicate with #38

closed 2023-03-26 by k323r