flucto-gmbh/motion-sensor-box #27: Create service to send data via MQTT to Contact's Elements for IoT broker

ahaselsteiner, 2022-02-23:
Motion sensor boxes are used in Uni Bremen's project WindIO.

In this project vibration data from wind turbines shall be sent via the MQTT protocol to a broker of the Contact Elements for IoT software.

An example script has been created that successfully sends data from MSB logfile to the broker: https://github.com/ahaselsteiner/mqtt-client-msb

The new service shall end live data to the MQTT broker.

This service will obviously only be activated on the MSBs that are used in the WindIO project.

ahaselsteiner, 2022-02-23:
My current idea is to implement this as a service, somewhat similar to msb_fusionlog. 
The service would use zmq to receive data, similar to here:

https://github.com/flucto-gmbh/msb_fusionlog/blob/c1eba8618c3aa08182f34fd38fb5a37cccb849fa/src/msb_fusionlog.py#L64-L73

Then the received data would be sent, smililar to here:

https://github.com/ahaselsteiner/mqtt-client-msb/blob/a6df50a1df00878d4a173795338c099f4a62e7a0/src/example.py#L117-L120

ahaselsteiner, 2022-02-23:
I talked with Aljoscha: Implementing this as a service as described above makes sense. I will listen to the "imu" and "gps" topics to gather acceleration and gps (which include timestamps) data.

k323r, 2022-05-11:
This hase successfully been implemented and I am closing this issue

closed 2022-05-11 by k323r