flucto-gmbh/motion-sensor-box #54: Reimplement attitude service

pacjac, 2023-03-22:
- [ ] Evaluate algorithms  
- [ ] Evaluate AHRS library: Data Streams? Performance? 
- [ ] Generate test data set

pacjac, 2023-03-23:
See #38

k323r, 2023-03-23:
@kawindme As mentioned in #26 , the imu features a digital motion processor, that should - in theory - be able to provide a orientation quaternion out of the box. The issue with this feature is, that the attitude estimation is done by a fpga, whose firmware is a proprietary, binay blob that is tedious to flash onto the imu. There are some examples out there of people, that have gotten it to work. Might be worth it to investigate.

k323r, 2023-08-11:
msb_attitude has been reimplemented with a complementary filter in #123

closed 2023-08-11 by k323r