flucto-gmbh/motion-sensor-box #142: wifi power savings - permanent disable

k323r, 2023-08-07:
https://raspberrypi.stackexchange.com/questions/96606/make-iw-wlan0-set-power-save-off-permanent

kawindme, 2023-08-07:
an alternative would be (at least for the boxes running network manager):
https://gist.github.com/jcberthon/ea8cfe278998968ba7c5a95344bc8b55

kawindme, 2023-08-07:
info on rc.local:
https://unix.stackexchange.com/questions/49626/purpose-and-typical-usage-of-etc-rc-local

kawindme, 2023-08-11:
on msb-110 and msb-107 i could not disable the power saving mode with the edit it `etc/rc.local`.
Instead I simplified the systemd solution proposed in that stackexchange post.

`sudo systemctl --full --force edit wifi_powersave_off.service`

insert:
```
[Unit]
Description=Set WiFi power save off
After=sys-subsystem-net-devices-wlan0.device

[Service]   
Type=oneshot
RemainAfterExit=yes
ExecStart=/sbin/iw dev wlan0 set power_save off

[Install]
WantedBy=sys-subsystem-net-devices-wlan0.device
```
`sudo systemctl enable wifi_powersave_off.service`
 then reboot.

`iw wlan0 get power_save` should report: `Power save: off`